switux/AstroWeatherStation #52: Unimplemented Endpoint

ispybadguys, 2023-12-20:
Great project! 

I am not sure what I am going wrong but when I try and connect using Ascom Diagnostics to with the dome or the observing conditions monitor I am getting:

[DEBUG] ALPACA: unimplemented endpoint: 8 /api/v1/observingconditions/0/connected, with parameters: (ClientTransactionID=1) (ClientID=49360) (Connected=True)

Any help is appreciated.

Kurt

switux, 2023-12-20:
Hello

Happy to see you like it! I need more details, can you please dump the contents of the boot time banner?

Thanks
F

ispybadguys, 2023-12-20:
I did change the datamancers.net website to a dummy because I didn't know exactly if this would cause problems. I also there are a few things that I didn't understand in the configuration but ASCOM DIagonistics does discover the device and tries to connect. At this point I do not actually have any external hardware attached.

22:50:40.259 -> ##############################################################################################
22:50:40.259 -> # AstroWeatherStation                                                                        #
22:50:40.259 -> #  (c) Lesage Franck - lesage@datamancers.net                                                #
22:50:40.259 -> #--------------------------------------------------------------------------------------------#
22:50:40.259 -> # MCU              : Model ESP32-PICO-D4 Revision 1                                          #
22:50:40.259 -> # WIFI Mac         : 64:b7:08:80:fb:b8                                                       #
22:50:40.259 -> # Power Mode       : 12V DC                                                                  #
22:50:40.259 -> # PCB version      : 3.x                                                                     #
22:50:40.259 -> # Anemometer model : PR-3000-FSJT-N01                                                        #
22:50:40.259 -> # Windvane model   : PR-3000-FXJT-N01                                                        #
22:50:40.259 -> # Ethernet present : Yes                                                                     #
22:50:40.259 -> # Firmware         : 3.0.0.0-20231124                                                        #
22:50:40.259 -> #--------------------------------------------------------------------------------------------#
22:50:40.259 -> # GPIO PIN CONFIGURATION                                                                     #
22:50:40.259 -> #--------------------------------------------------------------------------------------------#
22:50:40.259 -> # Anemometer : RX=33 TX=25 CTRL=32                                                           #
22:50:40.259 -> # Wind vane  : RX=26 TX=14 CTRL=27                                                           #
22:50:40.259 -> # RG9        : RX=17 TX=16 MCLR=15 RAIN=35                                                   #
22:50:40.350 -> # DEBUG      : 34                                                                            #
22:50:40.350 -> #--------------------------------------------------------------------------------------------#
22:50:40.350 -> # RUNTIME CONFIGURATION                                                                      #
22:50:40.350 -> #--------------------------------------------------------------------------------------------#
22:50:40.350 -> # AP SSID      : AstroWeatherStation                                                         #
22:50:40.350 -> # AP PASSWORD  : xxxxxxxxxx                                                               #
22:50:40.350 -> # AP IP        : 192.168.1.1/24                                                              #
22:50:40.350 -> # AP Gateway   : 192.168.1.1                                                                 #
22:50:40.350 -> # STA SSID     : Kurt's Wi-Fi Network                                                        #
22:50:40.350 -> # STA PASSWORD : xxxxxxxxx                                                               #
22:50:40.350 -> # STA IP       : 192.168.1.62/24                                                             #
22:50:40.430 -> # STA Gateway  : 192.168.1.1                                                                 #
22:50:40.430 -> # SERVER       : www.tempestwx.com                                                           #
22:50:40.430 -> # URL PATH     : /station/27577/                                                             #
22:50:40.430 -> # TZNAME       : MST7MDT                                                                     #
22:50:40.430 -> # ROOT CA      : -----BEGIN CERTIFICATE----- MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQY #
22:50:40.430 -> # JKoZIhvcNAQELBQAw TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh cmNoIEd #
22:50:40.430 -> # yb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4 WhcNMzUwNjA0MTEwNDM4WjBPMQswCQYD #
22:50:40.430 -> # VQQGEwJVUzEpMCcGA1UEChMgSW50ZXJu ZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgU #
22:50:40.430 -> # m9vdCBY MTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygc h77ct984kIxuPOZXo #
22:50:40.430 -> # Hj3dcKi/vVqbvYATyjb3miGbESTtrFj/RQSa78f0uoxmyF+ 0TM8ukj13Xnfs7j/EvEhmkvBioZxaUpmZmyPfjxwv6 #
22:50:40.430 -> # 0pIgbz5MDmgK7iS4+3mX6U A5/TR5d8mUgjU+g4rk8Kb4Mu0UlXjIB0ttov0DiNewNwIRt18jA8+o+u3dpjq+sW T8 #
22:50:40.430 -> # KOEUt+zwvo/7V3LvSye0rgTBIlDHCNAymg4VMk7BPZ7hm/ELNKjD+Jo2FR3qyH B5T0Y3HsLuJvW5iB4YlcNHlsdu8 #
22:50:40.732 -> # 7kGJ55tukmi8mxdAQ4Q7e2RCOFvu396j3x+UC B5iPNgiV5+I3lg02dZ77DnKxHZu8A/lJBdiB3QW0KtZB6awBdpUK #
22:50:40.732 -> # D9jf1b0SHzUv KBds0pjBqAlkd25HN7rOrFleaJ1/ctaJxQZBKT5ZPt0m9STJEadao0xAH0ahmbWn OlFuhjuefXKn #
22:50:40.732 -> # EgV4We0+UXgVCwOPjdAvBbI+e0ocS3MFEvzG6uBQE3xDk3SzynTn jh8BCNAw1FtxNrQHusEwMFxIt4I7mKZ9YIqio #
22:50:40.732 -> # ymCzLq9gwQbooMDQaHWBfEbwrbw qHyGO0aoSCqI3Haadr8faqU9GY/rOPNk3sgrDQoo//fb4hVC1CLQJ13hef4Y53 #
22:50:40.732 -> # CI rU7m2Ys6xt0nUW7/vGT1M0NPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNV HRMBAf8EBTADAQH/MB0GA1 #
[... cut ...]
22:50:40.732 -> # RH/TEMP/PRES : No                                                                          #
22:50:40.732 -> # WINDVANE     : No                                                                          #
22:50:40.732 -> # ANEMOMETER   : Yes                                                                         #
22:50:40.732 -> # RAIN SENSOR  : No                                                                          #
22:50:40.758 -> ##############################################################################################

switux, 2023-12-20:
That's weird, I cannot reproduce your issue and it just does not make sense ... may I ask that you try the version found in https://www.datamancers.net/images/AstroWeatherStation_3.0.0.tgz that contains a couple of additional traces and send me the complete debug log?

Also, what tool do you use to test the alpaca api (version, etc.) ? On my side, I use ConformU Version 2.0.0 (Build 16486))

You may have to change a couple of settings in aws.conf as this version contains other changes that are not committed yet on GH, like I said, it's still WIP :-)

Thanks

ispybadguys, 2023-12-20:
Thanks

Let me give that a try. What I know is that the route that the code is taking is directly to alpaca_server::does_not_exist via the server->onNotFound( std::bind( &alpaca_server::does_not_exist, this, std::placeholders::_1 ));.

This is all very new to me so I was using ASCOM Diagnostics to get access to the ASCOM Chooser because NINA does not use the Chooser.

What I see in the trace is this

09:50:46.348 GetBooleanValue           Include Debug Trace = 'True'
09:50:46.350 GetDecimalValue           Port Number = '11111'
09:50:46.350 GetDecimalValue           Remote Device Number = '0'
09:50:46.351 GetInt32Value             Establish Connection Timeout = '2'
09:50:46.352 GetInt32Value             Standard Device Response Timeout = '10'
09:50:46.352 GetInt32Value             Long Device Response Timeout = '120'
09:50:46.355 GetBooleanValue           Manage Connect Locally = 'False'
09:50:46.355 GetInt32Value             Image Array Transfer Type = '3'
09:50:46.356 GetInt32Value             Image Array Compression = '0'
09:50:46.357 GetBooleanValue           Enable Rediscovery = 'True'
09:50:46.358 GetBooleanValue           Enable IPv4 Discovery = 'True'
09:50:46.358 GetBooleanValue           Enable IPv6 Discovery = 'True'
09:50:46.359 GetInt32Value             Discovery Port = '32227'
09:50:46.359 GetBooleanValue           Trust Unsigned SSL Certificates = 'True'
09:50:46.359 ReadProfile 0             New values: Device IP: 192.168.1.62 11111, Remote device number: 0
09:50:46.359 Dome 0                    Trace state: True, Debug Trace State: True, TraceLogger Debug State: True
09:50:46.359 Dome 0                    Starting initialisation, Version: 6.0.0.0
09:50:46.359 Dome 25510                This instance's unique client number: 25510
09:50:46.359 Dome 25510                Connecting to device: 192.168.1.62:11111, Unique ID: Unknown through URL: http://192.168.1.62:11111
09:50:46.359 Dome 25510                Testing whether client at address http://192.168.1.62:11111 can be contacted.
09:50:46.360 ClientIsUp 25510          Created an InterNetwork TCP client
09:50:46.361 ClientIsUp 25510          Created connection task
09:50:46.361 ClientIsUp 25510          Created timeout task
09:50:46.361 ClientIsUp 25510          Waiting for a task to complete
09:50:48.369 ClientIsUp 25510          A task has completed
09:50:48.369 ClientIsUp 25510          Unable to contact client....
09:50:48.369 Dome 25510                The device at the configured IP address and port 192.168.1.62 cannot be contacted, attempting to re-discover it
09:50:50.391 Dome 25510                Found Alpaca device 192.168.1.62:8080 - AWS
09:50:50.391 Dome 25510                Found ASCOM device Rain sensor:Safetymonitor - be358e98-7cff-11ee-8804-93595b17357e at 192.168.1.62:8080
09:50:50.391 Dome 25510                Found ASCOM device Dome:Dome - e3a90ace-7cff-11ee-b0f6-73db65e0b158 at 192.168.1.62:8080
09:50:50.391 Dome 25510                Found ASCOM device AstroWeatherstation:ObservingConditions - ed0f1194-7cff-11ee-adf4-43fdf91ce20d at 192.168.1.62:8080
09:50:50.391 Dome 25510                Found ASCOM device Fake telescope:Telescope - 17e70650-8733-11ee-aa35-8f2796da5b48 at 192.168.1.62:8080
09:50:50.391                           
09:50:50.391 Dome 25510                No ASCOM device was discovered that had a UniqueD of Unknown
09:50:50.391                           
09:50:50.391 Dome 25510                ***** ALERT ***** UNSIGNED SSL CERTIFICATES WILL BE TRUSTED
09:50:50.391 Dome 25510                This devices's base URI: api/v1/Dome/0/
09:50:50.391 Dome 25510                Establish communications timeout: 2
09:50:50.391 Dome 25510                Standard device response timeout: 10
09:50:50.391 Dome 25510                Long device response timeout: 120
09:50:50.391 Dome 25510                User name: 
09:50:50.391 Dome 25510                Password is Null or Empty: True, Password is Null or White Space: True
09:50:50.391 Dome 25510                Password length: 0
09:50:50.391 Dome 25510                Completed initialisation
09:50:50.392 Connect 25510             Acquiring connection lock
09:50:50.392 Connect 25510             Has connection lock
09:50:50.392 IsClientConnected 25510   Number of connected devices: 0, Returning: False
09:50:50.392 Connect 25510             This is the first connection so set Connected to True
09:50:50.392 Connected 25510           Client Txn ID: 10, Sending command to remote device
09:50:50.392 Connected 25510           Client base URL: 'http://192.168.1.62:11111/', URI base: 'api/v1/Dome/0/', Method: Connected.
09:50:52.411 Connected 25510           Returned data content type: ''
09:50:52.411 Connected 25510           Status code: 0, Error message: 'The operation has timed out', Response status: Error
09:50:52.411 Connected 25510           Exception: System.Net.WebException: The operation has timed out

Kurt

ispybadguys, 2023-12-21:
Lesage

I feel bad about bugging you. I downloaded your 3.0.0 and after erasing flash and loading the config file I am seeing it work except for a strange error

21:38:04.380 -> [INFO] AstroWeatherStation is booting...
21:38:04.518 -> [DEBUG] Empty config file [/aws.conf]
21:38:04.518 -> [DEBUG] File successfuly read... and configuration format is valid... OK!
21:38:04.518 -> [INFO] Using minimal/factory config file.
21:38:04.518 -> [DEBUG] Booting in STA mode.
21:38:04.609 -> [INFO] Attempting to connect to SSID [Kurt's Wi-Fi Network] ... OK. Using IP [192.168.1.62]
21:38:07.621 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
21:38:07.621 -> 
21:38:07.621 -> Core  1 register dump:
21:38:07.621 -> PC      : 0x401981fb  PS      : 0x00060130  A0      : 0x800d7490  A1      : 0x3ffb21a0  
21:38:07.621 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3f400cad  A5      : 0x3ffb81e8  
21:38:07.621 -> A6      : 0x3ffb99dc  A7      : 0x3f4008c9  A8      : 0x800d52c0  A9      : 0x3ffb2180  
21:38:07.621 -> A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000017  A13     : 0x3ffb1e00  
21:38:07.667 -> A14     : 0x00000009  A15     : 0x3ffb2000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
21:38:07.667 -> EXCVADDR: 0x000000b8  LBEG    : 0x4008cc71  LEND    : 0x4008cc81  LCOUNT  : 0xfffffffc  
21:38:07.667 -> 
21:38:07.667 -> 
21:38:07.667 -> Backtrace: 0x401981f8:0x3ffb21a0 0x400d748d:0x3ffb21c0 0x400dc0cf:0x3ffb2260 0x400f85ba:0x3ffb2290

PC: 0x401981fb
EXCVADDR: 0x000000b8

Decoding stack results
0x401981f8: AWSSensorManager::get_i2c_mutex() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AWSSensorManager.cpp line 69
0x400d748d: AstroWeatherStation::initialise() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AWS.cpp line 451
0x400dc0cf: setup() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AstroWeatherStation_3.0.0.ino line 46
0x400f85ba: loopTask(void*) at C:\Users\Kurt\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.8\cores\esp32\main.cpp line 42

The code that is crashing is:

68: SemaphoreHandle_t AWSSensorManager::get_i2c_mutex( void )
69: {
70:	return i2c_mutex;
71: }

Thanks to your tip I discovered the ASCOM Comfort Tester which I will try.

My was.conf.dfl I changed to this which I think is run everything in WiFi STA mode.
{"pcb_version":"3.x","pwr_mode":2,"has_ethernet":0,"has_sc16is750":0,"pref_iface":1,"alpaca_iface":1,"config_iface":1}

Thanks

Kurt

switux, 2023-12-21:
Woops :-(

Corner case not caught in the code when you have no sensor attached. You can work around this bug but activating the RH/TEMP/PRES sensor.

I will open a specific bug for this.

No problem bugging me, it is by sharing code that we can improve it, that is the whole point.

Thanks for your cooperation,
Franck

ispybadguys, 2023-12-21:
Thanks I appreciate the help.

I set 

#define	DEFAULT_HAS_BME					1

even though I don’t have one connected. I suppose I could attach one. Am I missing something?

I still get the error 

Decoding stack results
0x401981fc: AWSSensorManager::get_i2c_mutex() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AWSSensorManager.cpp line 69
0x401981fc: AWSSensorManager::get_i2c_mutex() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AWSSensorManager.cpp line 69
0x400d748d: AstroWeatherStation::initialise() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AWS.cpp line 451
0x400d748d: AstroWeatherStation::initialise() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AWS.cpp line 451
0x400dc0d3: setup() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AstroWeatherStation_3.0.0.ino line 46
0x400dc0d3: setup() at Z:\Arduino 2022\AstroWeatherStation_3.0.0/AstroWeatherStation_3.0.0.ino line 46
0x400f85be: loopTask(void*) at C:\Users\Kurt\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.8\cores\esp32\main.cpp line 42
0x400f85be: loopTask(void*) at C:\Users\Kurt\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.8\cores\esp32\main.cpp line 42

BTW I love the hash function approach to your Case Statements using Char*  I have never seen this and it is great. I also am impressed by the RE use in the HTML route setup. I didn’t know tis was possible.

Kurt

> On Dec 21, 2023, at 5:29 AM, Lesage Franck ***@***.***> wrote:
> 
> RH

switux, 2023-12-21:
My mistake, I have been too hasty to move some code and did not run all the tests: capital sin ...

You must have enabled dome control and that explains the problem. You can download again the tgz, I have overwritten it.

About the code tricks, well ... I build on others' work :-) Happy if, in turn, it inspires others!

Cheers
F

ispybadguys, 2023-12-26:
Merry Christmas Franck

I got the last code set running for the OC server and I ran Conform but it fils for me.

20:54:41.771 ASCOM Universal Device Conformance Checker Version 2.0.0.16486, Build time: Mon 29 May 2023 08:03:52
20:54:41.772 
20:54:41.773 Operating system is Microsoft Windows 10.0.22631 64bit, Application is 64bit.
20:54:41.773 
20:54:41.774 Alpaca device: AstroWeatherstation (192.168.1.62:8080 ObservingConditions/0)
20:54:41.774 
20:54:41.779 CreateDevice                        INFO     Creating Alpaca device: IP address: 192.168.1.62, IP Port: 8080, Alpaca device number: 0
20:54:41.783 CreateDevice                        INFO     Alpaca device created OK
20:54:41.783 CreateDevice                        INFO     Successfully created driver
20:54:42.785 ConformanceCheck                    OK       Driver instance created successfully
20:54:42.785 
20:54:42.786 Connect to device
20:54:42.931 Connected                           ISSUE    Exception when setting Connected to True: Error calling method: Connected, HTTP Completion Status: BadRequest, Error Message:
Endpoint does not exist
20:54:42.945 
20:54:42.945 ConformanceCheck                             Further tests abandoned.
20:54:42.945 
20:54:42.946 Your device had 1 issue, 0 errors and 0 configuration alerts
20:54:42.946 
20:54:42.947 Issue Summary
20:54:42.947 Connected                           ISSUE    Exception when setting Connected to true - testing abandoned: Error calling method: Connected, HTTP Completion Status: BadRequest, Error Message:
Endpoint does not exist

Whe I was debugging the previous version it seems that the server->on route for the connected path was not getting taken. The Serial Monitor is showing 
20:54:40.411 -> [DEBUG] NTP Synchronised. Time and date: 2023-12-25 20:54:40
20:54:42.925 -> 
20:54:42.925 -> [DEBUG] ALPACA: [2] unimplemented endpoint: 8 /api/v1/observingconditions/0/connected, with parameters: (Connected=True) (ClientID=1) (ClientTransactionID=1) 
20:54:45.132 -> [DEBUG] Temperature = -0.52 °C
20:54:45.132 -> [DEBUG] Pressure = 820.42 hPa
20:54:45.171 -> [DEBUG] RH = 0.00 %

Kurt

ispybadguys, 2023-12-26:
Please excuse the reply to my reply I tried the following. I added a route

 server->on( "/api/v1/observingconditions/0/connected", 8, std::bind( &alpaca_server::dispatch_request, this, std::placeholders::_1 ));

and this route actually seems to catch the connected url.

I get this response in the Serial monitor 

21:21:44.354 -> [DEBUG] NTP Synchronised. Time and date: 2023-12-25 21:21:44
21:21:48.628 -> 
21:21:48.628 -> [DEBUG] Alpaca client request parameters: (Connected=True)(ClientID=1)(ClientTransactionID=1)
21:21:49.350 -> [DEBUG] NTP Synchronised. Time and date: 2023-12-25 21:21:49

but ASCOM Conform still complains

21:21:47.517 ASCOM Universal Device Conformance Checker Version 2.0.0.16486, Build time: Mon 29 May 2023 08:03:52
21:21:47.518 
21:21:47.518 Operating system is Microsoft Windows 10.0.22631 64bit, Application is 64bit.
21:21:47.518 
21:21:47.519 Alpaca device: AstroWeatherstation (192.168.1.62:8080 ObservingConditions/0)
21:21:47.519 
21:21:47.522 CreateDevice                        INFO     Creating Alpaca device: IP address: 192.168.1.62, IP Port: 8080, Alpaca device number: 0
21:21:47.526 CreateDevice                        INFO     Alpaca device created OK
21:21:47.526 CreateDevice                        INFO     Successfully created driver
21:21:48.528 ConformanceCheck                    OK       Driver instance created successfully
21:21:48.528 
21:21:48.528 Connect to device
21:21:52.128 Connected                           ISSUE    Exception when setting Connected to True: The response ended prematurely.
21:21:52.130 
21:21:52.130 ConformanceCheck                             Further tests abandoned.
21:21:52.130 
21:21:52.131 Your device had 1 issue, 0 errors and 0 configuration alerts
21:21:52.131 
21:21:52.131 Issue Summary
21:21:52.131 Connected                           ISSUE    Exception when setting Connected to true - testing abandoned: The response ended prematurely.

Kurt

switux, 2023-12-26:
Hello and merry christmas too!

The more I think about it, the more an issue with a library becomes a possibility, especially with the RE. Which version of ESPASyncWebSrv do you have?

Can you also list the libraries that got compiled?

BTW I pushed code yesternight on GH , not that it will solve your issue, but contains fixes for various bugs :-)

Cheers,
F

ispybadguys, 2023-12-26:
Franck

I have 1.2.7 installed author is dvarrel

I found where it is going by modifying this code:

void alpaca_server::dispatch_request( AsyncWebServerRequest *request )
{Serial.printf("alpaca_server::dispatch_request\n" );
	if ( !extract_transaction_details( request, ( request->method() != HTTP_GET ) )) {
    Serial.printf("alpaca_server::dispatch_request != HTTP_GET\n" );
		request->send( 400, "text/plain", transaction_details );
		return;
	}
  Serial.printf("alpaca_server::dispatch_request request->pathArg(0).c_str()=%s\n",request->pathArg(0).c_str() ); // KRL
	switch( str2int( request->pathArg(0).c_str() )) {

		case str2int( "dome" ):
			dispatch_dome_request( request );
			break;

		case str2int( "observingconditions" ):
      Serial.printf("DISPATCH OC\n" ); //KRL
			dispatch_observingconditions_request( request );
			break;

		case str2int( "safetymonitor" ):
			dispatch_safetymonitor_request( request );
			break;

		case str2int( "telescope", 0 ):
			dispatch_telescope_request( request );
			break;
    default:
      Serial.printf("alpaca_server::dispatch_request request->pathArg(0).c_str() matches none of the choices\n",request->pathArg(0).c_str() ); // KRL
	}
}

So it  is arriving at the default: Here is the Serial Monitor

07:23:31.694 -> [DEBUG] NTP Synchronised. Time and date: 2023-12-26 07:23:31
07:23:32.345 -> 
07:23:32.345 -> [DEBUG] Alpaca client request parameters: 
07:23:32.436 -> 
07:23:32.436 -> [DEBUG] Alpaca client request parameters: 
07:23:32.832 -> [DEBUG] Temperature = -3.11 °C
07:23:32.832 -> [DEBUG] Pressure = 820.49 hPa
07:23:32.832 -> [DEBUG] RH = 0.00 %
07:23:32.832 -> [DEBUG] Dew point = -nan °C
07:23:36.688 -> [DEBUG] NTP Synchronised. Time and date: 2023-12-26 07:23:36
07:23:40.286 -> alpaca_server::dispatch_request
07:23:40.286 -> 
07:23:40.286 -> [DEBUG] Alpaca client request parameters: (Connected=True)(ClientID=1)(ClientTransactionID=7)
07:23:40.286 -> alpaca_server::dispatch_request request->pathArg(0).c_str()=
07:23:40.339 -> alpaca_server::dispatch_request request->pathArg(0).c_str() matches none of the choices

I am fairly confident it took this route I created

server->on( "/api/v1/observingconditions/0/connected", 8, std::bind( &alpaca_server::dispatch_request, this, std::placeholders::_1 ));

Kurt

> On Dec 26, 2023, at 2:23 AM, Lesage Franck ***@***.***> wrote:
> 
> ESPASyncWebSrv

switux, 2023-12-26:
Kurt

*BONK*

I just finally found the root cause (I think) while reviewing ESPAsyncWebSrv docs, I had the exactly same problem a few weeks ago (geez, why 1/ did I not put a warning in the code? 2/ did it take so long for me to remember? ).

You need to add the following (pretty self-explanatory) line to the file named 'platform.local.txt', which is (for me), in ~/.arduino15/packages/esp32/hardware/esp32/2.0.14

**_compiler.cpp.extra_flags=-DASYNCWEBSERVER_REGEX=1_**

Chances are it will be the end of this problem :-)

Cheers
F

ispybadguys, 2023-12-26:
Thanks

I see a platform.txt here but not a  ‘platform.local.txt’. I assume that I just add that file there?

Kurt

ispybadguys, 2023-12-27:
Franck

I got it working! Thanks so much for the help!

07:45:54.908 ASCOM Universal Device Conformance Checker Version 2.0.0.16486, Build time: Mon 29 May 2023 08:03:52
07:45:54.909 
07:45:54.909 Operating system is Microsoft Windows 10.0.22631 64bit, Application is 64bit.
07:45:54.909 
07:45:54.909 Alpaca device: AstroWeatherstation (192.168.1.62:8080 ObservingConditions/0)
07:45:54.909 
07:45:54.909 CreateDevice                        INFO     Creating Alpaca device: IP address: 192.168.1.62, IP Port: 8080, Alpaca device number: 0
07:45:54.909 CreateDevice                        INFO     Alpaca device created OK
07:45:54.909 CreateDevice                        INFO     Successfully created driver
07:45:55.911 ConformanceCheck                    OK       Driver instance created successfully
07:45:55.911 
07:45:55.911 Connect to device
07:45:56.113 Connected                           OK       True
07:45:56.113 
07:45:56.117 Common Driver Methods
07:45:56.209 InterfaceVersion                    OK       1
07:45:56.211 Connected                           OK       True
07:45:56.288 Description                         OK       OpenAstroDevices AWS Environmental and SQM sensors / v1.0.0
07:45:56.355 DriverInfo                          OK       OpenAstroDevices AWS Environmental and SQM sensors driver / v1.0.0
07:45:56.425 DriverVersion                       OK       1.0
07:45:56.521 Name                                OK       OpenAstroDevices AWS Environmental and SQM sensors driver / v1.0.0
07:45:56.521 Action                              INFO     Conform cannot test the Action method
07:45:56.593 SupportedActions                    OK       Driver returned an empty action list
07:45:56.594 
07:45:56.594 Properties
07:45:56.694 AveragePeriod                       OK       0
07:45:56.773 AveragePeriod Write                 OK       Invalid Value exception generated as expected on set average period < -1.0
07:45:56.859 AveragePeriod Write                 OK       Successfully set average period to 0.0
07:45:56.935 AveragePeriod Write                 OK       Invalid value exception thrown when average period is set to 5.0, which is permitted by the specification
07:45:57.018 AveragePeriod Write                 OK       Successfully restored original average period: 0
07:45:57.312 CloudCover                          OK       0
07:45:57.412 DewPoint                            ISSUE    Unexpected error: 'n' is invalid within a number, immediately after a sign character ('+' or '-'). Expected a digit ('0'-'9'). Path: $.Value | LineNumber: 0 | BytePositionInLine: 44.
07:45:57.536 Humidity                            OK       0
07:45:57.537 DewPoint & Humidity                 ISSUE    One of Dew point or humidity is implemented and the other is not. Both must be implemented or both must not be implemented per the interface specification
07:45:57.624 Pressure                            OK       824.5
07:45:57.691 RainRate                            ISSUE    Unexpected error: The station has no rain sensor
07:45:57.764 SkyBrightness                       OK       0
07:45:57.852 SkyQuality                          OK       0
07:45:57.975 StarFWHM                            OK       Optional member returned a PropertyNotImplementedException error.
07:45:58.116 SkyTemperature                      OK       0
07:45:58.206 Temperature                         OK       1.1
07:45:58.270 WindDirection                       OK       0
07:45:58.356 WindGust                            ISSUE    Invalid value (below minimum expected - 0): -1
07:45:58.739 WindSpeed                           OK       0
07:45:58.739 WindSpeed                           OK       Wind direction is reported as 0.0 when wind speed is 0.0
07:45:58.739 
07:45:58.740 Methods
07:45:58.813 TimeSinceLastUpdateLatest           OK       0
07:45:58.887 TimeSinceLastUpdateCloudCover       OK       0
07:45:58.974 TimeSinceLastUpdateDewPoint         OK       0
07:45:59.064 TimeSinceLastUpdateHumidity         OK       0
07:45:59.064 DewPoint & Humidity                 OK       Dew point and humidity are both implemented per the interface specification
07:45:59.141 TimeSinceLastUpdatePressure         OK       1
07:45:59.217 TimeSinceLastUpdateRainRate         OK       1
07:45:59.288 TimeSinceLastUpdateSkyBrightness    OK       1
07:45:59.355 TimeSinceLastUpdateSkyQuality       OK       1
07:45:59.419 TimeSinceLastUpdateStarFWHM         OK       Optional member returned a MethodNotImplementedException error.
07:45:59.489 TimeSinceLastUpdateSkyTemperature   OK       1
07:45:59.556 TimeSinceLastUpdateTemperature      OK       1
07:45:59.624 TimeSinceLastUpdateWindDirection    OK       1
07:45:59.706 TimeSinceLastUpdateWindGust         OK       1
07:45:59.783 TimeSinceLastUpdateWindSpeed        OK       1
07:46:11.033 Refresh                             ISSUE    Unexpected error: Sensor is not connected
07:46:11.101 SensorDescriptionCloudCover         ISSUE    Unexpected error: Dome is not connected
07:46:11.161 SensorDescriptionDewPoint           ISSUE    Unexpected error: Dome is not connected
07:46:11.221 SensorDescriptionHumidity           ISSUE    Unexpected error: Dome is not connected
[... 54 lines cut ...]
07:46:11.974 SensorDescriptionRainRate           ISSUE    Unexpected error: Dome is not connected
07:46:11.974 SensorDescriptionSkyBrightness      ISSUE    Unexpected error: Dome is not connected
[... cut ...]
07:46:11.975 Consistency - WindDirection         ISSUE    Sensor value is implemented: True, Sensor description is implemented: False, Sensor time since last update is implemented: True
07:46:11.975 Consistency - WindGust              ISSUE    Sensor value is implemented: True, Sensor description is implemented: False, Sensor time since last update is implemented: True
07:46:11.975 Consistency - WindSpeed             ISSUE    Sensor value is implemented: True, Sensor description is implemented: False, Sensor time since last update is implemented: True

Hope you have a blessed New Year!

Kurt

switux, 2023-12-28:
Kurt

Happy to see it eventually works :-) I added a cpp directive to throw an error if the value is not set.

I can however see some issues from the conformance test. I opened a bug for one (wind gust = -1) but others, especially with the dew point is not clear for me, would you mind sharing the output you got from the station?

I will chase the other issues myself.

If you plan to add other sensors, please let me know, so that I can see if I want to add them to the baseline. It was originally a pet project but it went out of hands and I actually got orders (can you believe it?) ... other people may have different needs that I never contemplated.

Thanks
F

closed 2023-12-28 by switux

ispybadguys, 2023-12-29:
Franck

Do you want the output from the conform test or the output from the serial monitor or both.

I was modifying the code to get all of the sensor data from MQTT. I have a WeatherFlow Tempest Sensor which is pretty cool. It has an ultrasonic wind sensor and the data is available via REST or UDP.  You are a far better programmer than I so I would be honored to work with you. I am hosting the software on an M5Stack ESP32 which has a 5v compatible  i2c https://docs.m5stack.com/en/unit/extio for roof sensors and UPS sensors and a 4 relay unit for pier and roof control.  The i2c relay fixes problems where the roof would open on restart.

Kurt

⁣Get BlueMail for Android ​

ispybadguys, 2023-12-29:
Conform output 

14:19:20.391 ASCOM Universal Device Conformance Checker Version 2.0.0.16486, Build time: Mon 29 May 2023 08:03:52
14:19:20.392 
14:19:20.392 Operating system is Microsoft Windows 10.0.22631 64bit, Application is 64bit.
14:19:20.392 
14:19:20.393 Alpaca device: AstroWeatherstation (192.168.1.62:8080 ObservingConditions/0)
14:19:20.393 
14:19:20.396 CreateDevice                        INFO     Creating Alpaca device: IP address: 192.168.1.62, IP Port: 8080, Alpaca device number: 0
14:19:20.399 CreateDevice                        INFO     Alpaca device created OK
14:19:20.399 CreateDevice                        INFO     Successfully created driver
14:19:21.402 ConformanceCheck                    OK       Driver instance created successfully
14:19:21.403 
14:19:21.403 Connect to device
14:19:22.424 Connected                           OK       True
14:19:22.424 
14:19:22.428 Common Driver Methods
14:19:23.357 InterfaceVersion                    OK       1
14:19:23.359 Connected                           OK       True
14:19:23.745 Description                         OK       OpenAstroDevices AWS Environmental and SQM sensors / v1.0.0
14:19:24.030 DriverInfo                          OK       OpenAstroDevices AWS Environmental and SQM sensors driver / v1.0.0
14:19:24.401 DriverVersion                       OK       1.0
14:19:24.479 Name                                OK       OpenAstroDevices AWS Environmental and SQM sensors driver / v1.0.0
14:19:24.479 Action                              INFO     Conform cannot test the Action method
14:19:24.615 SupportedActions                    OK       Driver returned an empty action list
14:19:24.615 
14:19:24.615 Properties
14:19:24.715 AveragePeriod                       OK       0
14:19:24.799 AveragePeriod Write                 OK       Invalid Value exception generated as expected on set average period < -1.0
14:19:24.899 AveragePeriod Write                 OK       Successfully set average period to 0.0
14:19:25.020 AveragePeriod Write                 OK       Invalid value exception thrown when average period is set to 5.0, which is permitted by the specification
14:19:25.103 AveragePeriod Write                 OK       Successfully restored original average period: 0
14:19:25.203 CloudCover                          OK       0
14:19:25.292 DewPoint                            OK       -99
14:19:25.371 Humidity                            ISSUE    Invalid value (below minimum expected - 0): -1
14:19:25.371 DewPoint & Humidity                 OK       Dew point and humidity are both implemented per the interface specification
14:19:25.430 Pressure                            OK       0
14:19:25.496 RainRate                            ISSUE    Unexpected error: The station has no rain sensor
14:19:25.917 SkyBrightness                       OK       0
14:19:26.425 SkyQuality                          OK       0
14:19:26.810 StarFWHM                            OK       Optional member returned a PropertyNotImplementedException error.
14:19:26.883 SkyTemperature                      OK       0
14:19:26.979 Temperature                         OK       -99
14:19:27.038 WindDirection                       OK       0
14:19:27.099 WindGust                            ISSUE    Invalid value (below minimum expected - 0): -1
14:19:27.181 WindSpeed                           OK       0
14:19:27.181 WindSpeed                           OK       Wind direction is reported as 0.0 when wind speed is 0.0
14:19:27.181 
14:19:27.182 Methods
14:19:27.449 TimeSinceLastUpdateLatest           OK       0
14:19:27.564 TimeSinceLastUpdateCloudCover       OK       0
14:19:27.687 TimeSinceLastUpdateDewPoint         OK       0
14:19:27.818 TimeSinceLastUpdateHumidity         OK       0
14:19:27.819 DewPoint & Humidity                 OK       Dew point and humidity are both implemented per the interface specification
14:19:28.040 TimeSinceLastUpdatePressure         OK       1
14:19:28.107 TimeSinceLastUpdateRainRate         OK       1
14:19:28.332 TimeSinceLastUpdateSkyBrightness    OK       1
14:19:28.767 TimeSinceLastUpdateSkyQuality       OK       1
14:19:29.087 TimeSinceLastUpdateStarFWHM         OK       Optional member returned a MethodNotImplementedException error.
14:19:29.179 TimeSinceLastUpdateSkyTemperature   OK       2
14:19:29.257 TimeSinceLastUpdateTemperature      OK       2
14:19:29.336 TimeSinceLastUpdateWindDirection    OK       2
14:19:29.391 TimeSinceLastUpdateWindGust         OK       2
14:19:29.472 TimeSinceLastUpdateWindSpeed        OK       2
14:19:41.553 Refresh                             ISSUE    Unexpected error: Sensor is not connected
14:19:42.155 SensorDescriptionCloudCover         ISSUE    Unexpected error: Dome is not connected
14:19:42.884 SensorDescriptionDewPoint           ISSUE    Unexpected error: Dome is not connected
14:19:44.114 SensorDescriptionHumidity           ISSUE    Unexpected error: Dome is not connected
14:19:44.115 DewPoint & Humidity                 OK       Dew point and humidity are both not implemented per the interface specification
14:19:45.161 SensorDescriptionPressure           ISSUE    Unexpected error: Dome is not connected
[... 96 lines cut ...]
14:19:49.554 -> OC URL: [sensordescription]
14:19:50.238 -> 
14:19:50.238 -> [DEBUG] Alpaca client request parameters: (ClientID=1)(ClientTransactionID=51)(SensorName=Temperature)
14:19:50.271 -> DISPATCH OC
14:19:50.271 -> OC URL: [sensordescription]
14:19:50.408 -> 
[... cut ...]
14:19:51.124 -> [DEBUG] Alpaca client request parameters: (ClientID=1)(ClientTransactionID=54)(SensorName=WindSpeed)
14:19:51.124 -> DISPATCH OC
14:19:51.124 -> OC URL: [sensordescription]
14:19:51.794 -> 
14:19:51.794 -> [DEBUG] Alpaca client request parameters: (Connected=False)(ClientID=1)(ClientTransactionID=55)
14:19:51.794 -> DISPATCH OC
14:19:51.794 -> OC URL: [connected]
14:19:52.937 -> [INFO] Firmware OTA update result: No profile matches.
14:19:54.125 -> [DEBUG] NTP Synchronised. Time and date: 2023-12-29 14:19:54

switux, 2023-12-29:
Kurt,

Thanks for the output. I will soon push code correcting the "-1" values.

Now, regarding the weird "Dome is not connected" issue, I found the line but it is strange to get this because the device should be connected. The code will also fix the "Dome" part, it was an unfortunate copy/paste ... I can dig into this "not connected" issue once the code is out.

I am still validating a hardware design fix for the solar panel version. I will push all the code when it is ok. If everything is ok, this will be on this Sunday evening (for me).

Now, about your own version and additions, I had in mind of adding mqtt support at some point and I still plan to do it. However, I have three orders pending, two small ones requiring some stability validation and one big (according to my standards!) with quite some development ahead of me. That being said, the more, the merrier, some like to say :-) I would then be happy to integrate your code for the ultrasonic wind sensor and let you test it (as I do not have this device).

Cheers
F

ispybadguys, 2023-12-31:
Franck

I am almost there. I spend a day trying to get the code to compile. I added a set of functions to connect to MQTT and a function to parse the JSON data and fill in sensor_data_t with the values from the JSON. The code was failing because of the HTTP_DELETE problem in the AsysnWebServer library. I was able to get this to work but it is a HACK because what I did was add include all of the include files to the astroweather.ino file. Don't ask me why this fixed it but I tested adding the various libraries to the AsynWebServer Example program to see if I could break that and when all of them were added it still worked so I copied them to the astroweather.ino and magically it worked. You probably have better insight to this. 

A crucial thing I just don’t understand is how the Arduing IDE functions when the files are .cpp. Adding my four files to the code as .h and .cpp just didn’t work and the IDE seems to do strange things like process my .h files more than once even thought is if #ifdef to prevent that . I just gave up and added these as .ino files and it compiles. 

What I need to do make the parsing of the JSON table driven so as a user you can map you json values to ASCOM OC values. the current coed has constructs like this to parse the JSON:
  if (doc.containsKey("temperature"))                        {sensorPtr->temperature     = (doc["temperature"].as<double>()-32.0)*5./9.;}

I’ll work on this a bit and then share it and maybe you can integrate it in a manner that isn’t a HACK.

Kurt

ispybadguys, 2023-12-31:
Franck

I had a thought to add to my previous email. I was wondering if you could incorporate an option to include user code for the OC, Dome, and Safety monitor that would simplify the integration of custom code? For example I am thinking I want to source the safety monitor from MQTT and incorporate my ROR control code.

I am building an integrated observatory control system in Home Assistant that allows me to gather weather data and other data from a large variety of sources and build simple graphical displays and incorporate controls. I allows you to subscribe and publish to MQTT and control almost any variety of power line switches. Here is a sample of one dashboard the displays mount data , my All-Sky camera, and some Blink cameras.

Kurt

￼

switux, 2024-01-01:
Kurt,

I took steps to modularise sensors, we could add a dedicated sensor class (like I recently did for the hydreon sensors) which we could enable/disable from the web ui like with any other sensor. You would then have to implement a small set of public methods (begin, initialise, read_sensor, for instance) which the sensormanager would use to initialise and read data, making a seamless integration.

You can fork my code on GH, add your code and then make a pull request so that I can review it and merge with mine.

If it is ok for you, I will make a 3.1.0 branch for you to fork, and track work on this branch. That way, you can experiment on your fork and when you are happy with the results, you pull request and I merge in my 3.1.0 and if it's ok I can merge to main once the 3.0.0 is released.

In the meantime, I will regularly merge 3.0.0 into 3.1.0 so that you can in turn pull recent code.

BTW, your attachement did not work (dashboard).

Cheers
F

ispybadguys, 2024-01-01:
Franck

That sounds like a plan. I'll need to study git a bit.

Kurt

⁣Get BlueMail for Android ​

switux, 2024-01-02:
Kurt

I suggest we continue this discussion in the "Discussions" tab. I created a topic for this.

F

ispybadguys, 2024-01-03:
Here is part of my HA dashboard that didn’t come thru.￼

switux, 2024-01-03:
I closed a few bugs related to alpaca protocol issues and some are related to the ones you had

commit 766e6a4b (2023-12-30): Close #52, cpp to throw error if ASYNCWEBSERVER_REGEX is not defined