switux/AstroWeatherStation #128: Unsafe Conditions

bucks0042, 2024-02-10:
Please consider to implement following unsafe condition :

Unsafe if :

- Cloudly for more as x seconds (To be defined in user interface)
- Overcast -> Immediately
- Rain -> Immediately
- wind : more than x seconds (TBDF in user interface) if wind is more as x m/s, immediately in case of wind more as y m/s

Revert to safe :

- immediately for cloudy to clear
- after x seconds for overcast
- after x seconds for Rain
- immediately for wind "x" after x seconds for wind "y"

switux, 2024-02-18:
[image]

switux, 2024-02-19:
It kind of works but there are some corner cases to be sorted out. I suggest we agree on a (simple) state machine asap.

commit d79c933e (2024-02-23): Close #135, close #125, close #128, close #103, close #124
commit e16707ea (2024-02-26): Finalised lookout and improved UI (#138)

* Close #135, close #125, close #128, close #103, close #124
closed 2024-02-23 by switux